MangoAutomation/ma-dashboards #323: Data Point Bulk Editor Broken

terrypacker, 2020-03-11:
Not sure when this crept in but it was semi recently.  When I go to bulk edit any data point group I get this in the dev tools console and the inputs are not filled in and the save/cancel etc buttons are not shown.

```
RangeError: Maximum call stack size exceeded
    at Function.[Symbol.hasInstance] (<anonymous>)
    at mangoUi~ngMango~ngMangoServices.js?v=1996dc3bc5e2468551ca:57
    at Set.forEach (<anonymous>)
    at Function.combineInto (mangoUi~ngMango~ngMangoServices.js?v=1996dc3bc5e2468551ca:57)
    at mangoUi~ngMango~ngMangoServices.js?v=1996dc3bc5e2468551ca:57
    at Set.forEach (<anonymous>)
    at Function.combineInto (mangoUi~ngMango~ngMangoServices.js?v=1996dc3bc5e2468551ca:57)
    at mangoUi~ngMango~ngMangoServices.js?v=1996dc3bc5e2468551ca:57
    at Set.forEach (<anonymous>)
    at Function.combineInto (mangoUi~ngMango~ngMangoServices.js?v=1996dc3bc5e2468551ca:57)
```

terrypacker, 2020-03-11:
I have merged in some changes around log file viewing lately but I cannot confirm if this problem existed before those changes so I figure I'll let you take a look at this first @jazdw

terrypacker, 2020-04-03:
@pierpuccini I suspect this is now fixed but can you confirm this by bulk editing various points from various types of data sources.  Just sort of poke around a little.

pierpuccini, 2020-04-05:
Closing per coment in infiniteautomation/ma-modules-private#92

closed 2020-04-05 by pierpuccini