MangoAutomation/ma-dashboards #340: First page open with help out the home page format is bad

terrypacker, 2020-05-01:
Startup Mango on a fresh database to see the format.  The layout needs improvement.

pierpuccini, 2020-05-07:
[image]

jazdw, 2020-05-07:
@terrypacker There isn't much we can do about this besides popping the help over the top of the page. The page is responsive but responsive design is based upon the screen width. Opening the help does not change the screen width therefore the page does not adapt. We could detect if the help is open and change the styles but then you get a myriad of possible layouts -
i.e.
- help closed, menu closed, screen size less than 800px
- help closed, menu open, screen size less than 800px
- help open, menu closed, screen size less than 800px
- help open, menu open, screen size less than 800px
- help closed, menu closed, screen size less than 1024px
- help closed, menu open, screen size less than 1024px
- help open, menu closed, screen size less than 1024px
- help open, menu open, screen size less than 1024px

...you get the idea

terrypacker, 2020-05-08:
@Joelhaggar just making you aware as this issue was adding on your behalf.

terrypacker, 2020-05-08:
@jazdw and @pierpuccini I noticed that this is not specific to the home page, this is pretty much any page when you open help.  I would suggest that you pop it open over the page (or a new window/tab) if possible.

pierpuccini, 2020-05-08:
> @jazdw and @pierpuccini I noticed that this is not specific to the home page, this is pretty much any page when you open help. I would suggest that you pop it open over the page (or a new window/tab) if possible.

Regarding the help popping over I’m not sure if its the best way to go, because then the user wouldn't be able to interact with the UI he wants help on.