MangoAutomation/ma-dashboards #342: Data Source Edit - Lazy load data points

terrypacker, 2020-05-01:
I believe the reason that the points were not lazily loaded was due to the JSON export functionality but I'm not 100% sure on that.  Ideally we can make that list lazily load.

If I am correct on why it is not lazily loaded now I am suggest that we create a new data source export endpoint that allows exporting with points.  This will allow using RQL to get SeroJson files back.

jazdw, 2020-05-04:
Need metrics
- How long did the request take? Too long?
- Did the browser crash?
- How many data points did you have on the data source?

It was done this way for the bulk data point editing.

terrypacker, 2021-06-30:
Moved to Jira: https://radixiot.atlassian.net/browse/RAD-1025

closed 2021-06-30 by terrypacker