MangoAutomation/ma-dashboards #348: Publisher table bug - deleting publisher does not remove it from the list

terrypacker, 2020-05-07:
Starting with 0 publishers add 1 then try to remove it.  Still in list but gone from backend on refresh.  The same thing happens if I create 2 and then try to delete them.  Not sure if this is specific to PersistentTCP or not but that is what I tested with.

pierpuccini, 2020-05-07:
@terrypacker  I created all other types of publishers and they deleted as they should have, the only one I couldn't test is the bacnet one

jazdw, 2020-05-07:
@terrypacker It is likely a WebSocket issue, please test again and monitor your WebSocket to ensure the message gets delivered.

pierpuccini, 2020-05-07:
@terrypacker when monitoring the WebSocket, I don't get the notification for when the persistent TCP is deleted.

terrypacker, 2020-05-08:
Thanks guys, was a dao bug and is fixed.

closed 2020-05-08 by terrypacker