MangoAutomation/ma-dashboards #356: Make view script log file link able to be opened in new tab/window.

terrypacker, 2020-08-26:
This button doesn't allow you to open it in a new window which is painful:

[image]

This would apply to anywhere this is used, I assume it is a component but could be wrong.

jazdw, 2020-08-26:
Where is this? Meta point?

terrypacker, 2020-08-26:
Yes this is a meta point but would be the same for Scripting data sources and Email/Set Point event handlers I believe.

I also just noticed the link is wrong too:  My meta point is id 68 but the link takes me to: `...\logging-console?filenameFilter=metaPoint-4.log&filename=metaPoint-4.log`. 

It should be

`logging-console?filenameFilter=metaPoint-68.log&filename=metaPoint-68.log`

jazdw, 2020-09-01:
@terrypacker it retrieves the log file name from this end point using the data source XID - `/rest/latest/meta-data-source/log-file-path/DS_XID`. If this is not correct, please re-open. PS this was not my handiwork.

terrypacker, 2020-09-01:
@jazdw I will work this out with @pierpuccini, it was his handiwork anyway.

terrypacker, 2020-09-01:
@pierpuccini the REST endpoint was misleading.  It expects you to supply the XID of the data point (not the data source).  This is the problem.  The endpoint is now working correctly but the UI needs fixed can you:

- [ ] Open the link in a new window (or allow that by right clicking)
- [ ] Use the data point XID in the REST request to build the link

terrypacker, 2020-09-01:
@pierpuccini this issue is only relevant to Meta Data Points.

jazdw, 2020-09-01:
I already fixed most of it up in the commit above, this commit changes it to use the data point XID.
infiniteautomation/ma-modules-private@40c74fae30b9082e80a3ee3b0521f802e883bffc

commit 388951cf (2020-09-01): Fix log file view component

Remove unnecessary service
Closes #356
closed 2020-09-01 by jazdw