MangoAutomation/ma-dashboards #362: Permission Editor - Shows XIDs in some situations

terrypacker, 2020-10-21:
Have 3 roles that do exist I can see that in the "Other roles with access" view they display their XIDs and when displaying a newly added role without refreshing the page it displays the XID in the column in the table instead of the name.  This can be replicated by adding a new role while upgrading that is assigned to a permission on the item.  Here are some screenshots:

[image]
[image]
[image]

jazdw, 2021-03-25:
@terrypacker I have fixed up a couple of things around this but I am not sure what you mean in your steps to reproduce. Please test then close.

terrypacker, 2021-03-30:
@jazdw looks good.  One last thing if you feel it is necessary.  If I have 2 tabs open, one on Excel Reports (or anywhere with the permissions component) and I modify the name of an existing role that is being displayed (used) on the other tab open to the roles page.  The name is not modified in the column for that role on the permissions component.  However it is updated in the (+ Other roles with access) area.  It seems this is all happening via web socket so I figure the displayed role should also be updated to read the other name.

Not a big deal but we are here and working on this so figured its worth looking at.

jazdw, 2021-03-30:
@terrypacker I cannot reproduce the above, it updates in both places for me.

jazdw, 2021-03-31:
Reproduced problem with Terry's help, two more commits
4f7e5658f25e4bd8130d3508be036786ba22f51c
f728a34475407ad9f24d845a0e0df22e6eff7556

closed 2021-03-31 by jazdw