MangoAutomation/ma-dashboards #374: Event Handler Created with wrong referenceId2

terrypacker, 2021-02-03:
On the event handlers edit view on the event types tab if you select the upper checkbox for a data point event (may be others too) the referenceId2 field is sent as 0.  If you directly select the lower checkbox it works fine.

[image]

jazdw, 2021-02-08:
@terrypacker this is by design. You can choose for an event handler to handle all events for a data point or only a single detector.

closed 2021-02-08 by jazdw

terrypacker, 2021-02-08:
@jazdw I don't think this is wired up to work that way in the backend is it?

jazdw, 2021-02-12:
@terrypacker yes it should be. Re-open if not.

terrypacker, 2021-02-16:
Yes it is wired up, here is the dao method that handles it:

https://github.com/infiniteautomation/ma-core-public/blob/07ba5f25bc75ee8e0700e655d1ff381d2a2cf5e1/Core/src/com/serotonin/m2m2/db/dao/EventHandlerDao.java#L170